rppicomidi/midi2usbhost #1: USB MIDI host via built-in USB-C

jkuss, 2022-04-19:
I have a pimoroni lipo Pico which runs off of battery. This frees up the USB-C port. Is it possible for me to modify this project to send over usb? Apologies for the "issue".

rppicomidi, 2022-04-26:
@jkuss I am not familiar with the Pimoroni Pico LiPo, so I did a quick lookup on the Pimoroni website. I see no good reason the software won't "just work" as is. The UART pins are the same on the Raspberry Pi Pico board and the Pico LiPo. Unfortunately you will not be able to power your project from the LiPo battery using the Pimoroni board as is. You can hook up a separate 5V source to VBus pin on the Pimoroni Pico LiPo board. This will provide the 5V VBus a USB Host needs to supply downstream to the USB Device, and it will charge the battery. You will still need to adapt the USB C connector to attach to your USB MIDI device. This is probably not the solution you want.

You can probably hack the board to using a new USB breakout cable and a Step-up power supply to drive the USB Host VBus directly from the LiPo battery. It will be tricky soldering and you will have to be careful about protecting your board and your USB device from bad power states.

Sorry I am not more help.

rppicomidi, 2022-07-25:
@jkuss Did I answer your question here? May I close this issue?

closed 2022-07-25 by jkuss