rppicomidi/midi2usbhost #2: Using raspberry pi pico as a MIDI controller

krickovicd, 2022-05-19:
Hi. Sorry for reporting an issue but I don't see a discussion option.
I am trying to build a midi controller so that I can send midi commands to my NUX MG-300 guitar processor. I am currently using Arduino Uno for testing purposes and I have managed to send and receive messages (cc and sysex) going through PC. The problem is that arduino uno cannot be a usb host so I cannot connect MG-300 directly to it. There is an option of using a usb host shield but it is not cost effective and I think I will need more resources to implement everything that I envisioned.
I think you mentioned here https://github.com/hathach/tinyusb/discussions/1152 that you have an issue receiving messages from usb device. Is this resolved?
Do you think that this project can be modified so that I can send and receive MIDI messages to the MG-300?
I don't need the "old school MIDI IN and MIDI OUT" if it makes any difference.

rppicomidi, 2022-07-11:
@krickovicd Sorry I did not notice your post until now. The issue receiving from a midi device is resolved. Yes you can send and receive MIDI messages using the forked tinyusb on the midihost branch. Take a look at the MIDI host example code. The file  [midi_app.c](https://github.com/rppicomidi/tinyusb/blob/midihost/examples/host/midi/src/midi_app.c) shows sending and receiving messages directly. You can use this midi2usbhost project as another example. Make sure you are using my fork of tinyusb and check out the midihost git branch. Look at the [README](https://github.com/rppicomidi/midi2usbhost/blob/main/README.md#setting-up-your-build-and-debug-environment) section on how to set up your build environment if you don't know what I am talking about.

krickovicd, 2022-07-12:
Hi @rppicomidi, thank you for taking time to answer my question. I followed the instructions in the README section but failed to build the project. The first issue comes after executing "git checkout -b midihost origin/midihost" in the pico-sdk folder. After executing that command some files are deleted which makes cmake think that the pico-sdk is not installed in that folder anymore.
The next issue is that the "git clone --recurse-submodules https://github.com/midiusb2host.git" command is incorrect. As pointed in https://github.com/rppicomidi/midi2usbhost/issues/3#issuecomment-1181306369 it should be https://github.com/rppicomidi/midi2usbhost.git (at least I think so). Last thing, looking at the https://github.com/rppicomidi/tinyusb/tree/master/src/class/midi i don't see midi_host.h there.

rppicomidi, 2022-07-20:
Hi @krickovicd I have updated the README file. See the comments on the other issue. I corrected typos and hopefully clarified how to install the forked tinyusb library.

krickovicd, 2022-07-20:
Thanks @rppicomidi. Updated readme helped. I finally manged to build the project. Now I need to figure out how to communicate with my MG-300. I will check the examples you mentioned, hope that will be enough.
Thanks!

closed 2022-07-20 by krickovicd