rppicomidi/midi2usbhost #3: Pico PIO USB Host

Colatino, 2022-06-20:
Hi!
Not really an issue but arduino-pico, tinyUSB and Adafruit-tinyUSB just got updated to oficially support USB hosting with PIO on the RP2040. 
Maybe that can be a shortcut to get MIDI hosting working.

rppicomidi, 2022-06-26:
Thanks @Colatino Using PIO for USB is very interesting. However, if you use the code as described in this project with my forked tinyusb, MIDI hosting is working. It uses the built-in USB IP, not PIO. You can't hook the MIDI device to the Pico host through a hub with this code though. I am waiting for my pull request to the tinyusb  project to be merged to the tinyusb master branch. There has been a bit of activity on the host side of that project recently, so maybe my code will get merged eventually.

fadh3r, 2022-07-02:
Same here - could not compile with your instructions...

fadh3r, 2022-07-02:
I've got an error #include "class/midi/midi_host.h"

AshuSharda, 2022-07-07:
> https://github.com/csBlueChip/tinyusb/tree/master/src/class/midi take a look at this file with this got almost 90 % of compilation but then getting error please let me know if you are able to get it to work.

rppicomidi, 2022-07-08:
@fadh3r Please see my comments on issue #4 @Colatino I have never used the Arduino toolchain so I cannot comment.

fadh3r, 2022-07-12:
@rppicomidi by your readme.md I've got an error on stage:
'git clone --recurse-submodules https://github.com/midiusb2host.git'
with error
'Cloning into 'midiusb2host'...
remote: Not Found
fatal: repository 'https://github.com/midiusb2host.git/' not found
'''
so, this typo is fixing by adding right link to your github profile(name is missing in current command line)
'https://github.com/rppicomidi/midi2usbhost'

and I'm getting many different errors with include using current pico-sdk...

fadh3r, 2022-07-12:
```
[cmake] Using board configuration from /home/jj/_last_Pico_rppicomidi/pico-sdk/src/boards/include/boards/pico.h
[cmake] CMake Error at /home/jj/_last_Pico_rppicomidi/pico-sdk/src/CMakeLists.txt:23 (add_subdirectory):
[cmake]   add_subdirectory given source "tinyusb" which is not an existing directory.
[cmake] Call Stack (most recent call first):
[cmake]   /home/jj/_last_Pico_rppicomidi/pico-sdk/src/rp2_common/CMakeLists.txt:58 (pico_add_subdirectory)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
[cmake] See also "/home/jj/_last_Pico_rppicomidi/midi2usbhost/build/CMakeFiles/CMakeOutput.log".
[cmake] See also "/home/jj/_last_Pico_rppicomidi/midi2usbhost/build/CMakeFiles/CMakeError.log".
```

rppicomidi, 2022-07-20:
@fadh3r Thank you for finding typos and omissions in the build instructions. Please try the latest README.md file. As you pointed out, the path to the project code was wrong. More important, the instructions for using the forked tinyusb should have made clear you have to set the current directory to pico-sdk/lib/tinyusb first before you checkout the origin/midihost branch. Finally, I added instructions for a command line build. It assumes that the toolchain is installed correctly. I hope that helps.

fadh3r, 2022-07-22:
I'm very appreciated to you, @rppicomidi! I'll try it! Thanks!

MickeyDelp, 2022-07-22:
@rppicomidi Thanks for the doc updates. I've got it working now too. 
One thing I notice is that when a USB device is not connected, there is a constant stream of garbage MIDI out of the UART. Are you seeing that too or is it just my setup?

rppicomidi, 2022-07-25:
@MickeyDelp I don't have this setup connected at the moment. I don't recall this being an issue. May I ask that if the problem persists that you file a new Issue here? If others are seeing the same issue they might find the solution.

rppicomidi, 2022-07-25:
@Colatino Are you able to build the code now that I corrected typos and omissions in the README.md file? If so, do you mind if I close this issue? Adding MIDI Host support to PIO USB is a different project in my opinion.

Colatino, 2022-07-25:
I couldn't build it but my problem is the toolchain... for some reason it doesn't install properly.

You can close the issue, I opened it with the intent to sugest a new feature to the project but I'm a github noob and don't know how to.

MickeyDelp, 2022-07-26:
> @MickeyDelp I don't have this setup connected at the moment. I don't recall this being an issue. May I ask that if the problem persists that you file a new Issue here? If others are seeing the same issue they might find the solution.

It was just a noisy connection. I solved it by using a series diode on the UART TX. I would set that pin to open drain for MIDI, but the Pico does not seem to have that option.

rppicomidi, 2022-07-26:
@MickeyDelp I am glad your issue is solved. I am going to close this issue.

closed 2022-07-26 by rppicomidi

fadh3r, 2022-07-26:
Builded successfully!
Thank you @rppicomidi

rppicomidi, 2022-09-03:
@Colatino  I reopened this issue to let you know I did in fact implement PIO USB MIDI Host and generated a new project in my repo called [pico-usb-midi-filter](https://github.com/rppicomidi/pico-usb-midi-filter), which uses the PIO USB MIDI host and the built-in USB MIDI port as a MIDI device to create a MIDI message filter. The README for that project is hopefully enough better that you are able to built that project.  Thank you for calling the Pico-PIO-USB project to my attention.

Colatino, 2022-09-03:
That's great! I'll have a look as soon as possible.

AndrewCapon, 2022-11-11:
> Thanks @Colatino Using PIO for USB is very interesting. However, if you use the code as described in this project with my forked tinyusb, MIDI hosting is working. It uses the built-in USB IP, not [...]

Hi There,

Why is there the restriction to not using hubs, is this a TinyUSB thing?

Many thanks for any info

Andy

rppicomidi, 2022-11-15:
@AndrewCapon There is a bug somewhere in my tinyusb fork's implementation of the MIDI Host code that breaks hubs, I believe. However, there is a new midihost tinyusb pull request #1627 based on my code and a better implementation of USB host bulk endpoint support. Maybe @atoktoto got it working. I haven't had time to test it yet.

AndrewCapon, 2022-11-15:
Hi @rppicomidi,

Thanks for the info. 

I have tried the @atoktoto one as well, it has the same issue unfortunately.

I will take it up on #1627.

All the best, and thanks for the MIDI hosting code.

Andy

AndrewCapon, 2022-11-15:
Thanks for the info, If I get some time I will have a look, maybe another set of eyes may see something.

I had a quick look and from what I remember there is an enumeration for the hub, then for the first device plugged into the hub, but from that point on if you add a device it doesn't enumerate.

Also the disconnect only happens if you remove the hub, removing the device from the hub doesn't fire this.

Does this sound about right?

rppicomidi, 2022-11-17:
@AndrewCapon Yes, that's about right. It's weird because hubs work fine for MSC, which is not all that different from MIDI in that they both use 2 bulk endpoints.