rppicomidi/midi2usbhost #5: Add dual role / switchable mode

DatanoiseTV, 2022-10-31:
The Raspberry Pico and my boards have the GPIO24 which can detect if VBUS is coming from USB or not. Therefore, it can detect if the device is used as USB Host or Device.

Would be great to see dual mode / switched mode depending on the state of GPIO24.

rppicomidi, 2022-11-02:
@DatanoiseTV This looks technically feasible. What is the practical application? Do you want to be able to use the same device as a computer MIDI interface and a DIN MIDI to USB Device MIDI bridge?

rppicomidi, 2022-11-02:
@DatanoiseTV I am afraid I got caught up in what sounded like a great idea. Sadly, this is not technically feasible as you described it. There are two scenarios: in USB Device Mode, the Pico gets power from the USB host VBus pin; GPIO24 is pulled high because VBus is 5V. In USB Host Mode, the Pico has to supply 5V power to the VBus pin from an external supply, which also pulls GPIO24 high. It would be possible to make this manually switchable using a different GPIO and an external switch or jumper.

DatanoiseTV, 2022-11-04:
Well, it would work in a scenario where the board would be supplied via a LiPo to VSYS and VBUS would be used for a LiPo charging circuit, which would imply Device Mode and Host Mode if the LiPo is supplying VSYS (would require a Mosfet switching VBUS of the USB Host).

Anyhow, I think it would be awesome to make it change the mode between device and host by asserting a GPIO on Boot (or making it a software flag, perhaps in an emulated "EEPROM" on the device's flash. I have a lot of applications where this would be very useful.

rppicomidi, 2022-11-04:
@DatanoiseTV Well if you want it to poll a GPIO on boot to decide what to do and reboot if that GPIO changes, that should be straightforward. If you want to get fancy with a program flash stored setting to decide how the device behaves on startup, that is doable but less simple and requires more hardware to switch Vbus on and off. Either way, it is a different project from this one; not really an issue for the midi2usbhost project. However, if you want to have a go, I have some thoughts.

There is already a midi2usbdev project. Merge that project with this one. Instead of calling tusb_init(), just call tud_init() or tuh_init() depending on whether you need host or device mode, add tinyusb_device library to the target_link_libraries in the CMakeLists.txt file, and combine the tusb_config.h files from the two projects so you compile both host mode and device mode. A simple switching example would use a DPDT switch and a GPIO with the pull-up resistor enabled. Wire the center terminal of one pole of the switch to the GPIO and the center terminal of the other pole to Vbus. For the host switch position, connect the Vbus pole to 5V and wire the corresponding GPIO poll to ground. Then on startup, poll the GPIO. If it is low, then call tuh_init() and use the main loop from the midi2usbhost project; otherwise, call tud_init() and use the main loop from the midi2usbdev project. Poll the GPIO during normal operation. If it changes, just call watchdog_reboot(), and the system should behave as it would on power up. Of course, you have some issues. You need a connector that does host or device mode, and you should do some circuit protection work so your Pico board or power supply does not fry your host or get fried by the host if your gadget is switched to host mode whilst connected to a host.

Does that help? There are probably details missing here, but it will get you started.

DatanoiseTV, 2022-11-04:
I've been trying to adopt the code to include both TUH and TUD. Unfortunately, when linking (and including tinyusb_device and tinyusb_host), it fails to find references to various calls - perhaps it is an issue with using TUH+TUD together?

DatanoiseTV, 2022-11-04:
In fact, weirdly enough after adding tinyusb_device to the CMakeLists.txt and adopting the source code (incl adding the header for MIDI device),
it doesn't even know tud_task();

```c
sr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: CMakeFiles/midi2usbhost.dir/midi2usbhost.c.obj: in function `main':
midi2usbhost.c:(.text.startup.main+0x7a): undefined reference to `tud_task'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: midi2usbhost.c:(.text.startup.main+0x112): undefined reference to `tud_midi_n_available'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/midi2usbhost.dir/build.make:1303: midi2usbhost.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:1416: CMakeFiles/midi2usbhost.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
``

DatanoiseTV, 2022-11-04:
I wonder if this (found in the tinyusb dual role example) might be related.
Maybe it is not implemented yet.

```cmake
if (FAMILY STREQUAL "rp2040" AND NOT TARGET tinyusb_pico_pio_usb)
    message("Skipping dual host/device mode examples as Pico-PIO-USB is not available")
else()
```

rppicomidi, 2022-11-04:
Hmm, I don't know. I was able to use both modes in the [pico-usb-midi-filter](https://github.com/rppicomidi/pico-usb-midi-filter) project. The main difference there was I was using the Pico-PIO-USB project for the host USB hardware. Perhaps a look at that project will help.

DatanoiseTV, 2022-11-05:
I have submitted an Issue to the TinyUSB Git repo about that. After studying the RP2 Datasheet, I cannot see a reason why it could not be implemented with the real USB PHY, but only with PIO. I would prefer to go the USB PHY way, because PIO USB is of course not compliant with the USB phy standards and does 1.1 only.

rppicomidi, 2023-07-06:
@DatanoiseTV Given you submitted an issue to the TinyUSB git repo, I am closing this issue here.

closed 2023-07-06 by rppicomidi