rppicomidi/midi2usbhost #7: Problems with devices with multiple virtual cables

DatanoiseTV, 2022-11-21:
When using a Novation 24SL MKII for example, which presents itself as a 4 virtual cable device, the device will not work and report:

```
WARN: starting new transfer on already active ep 0 in
WARN: startassertion "!(buf_ctrl & USB_BUF_CTRL_FULL)" failed: file "/home/xxx/dev/pico-sdk/lib/tinyusb/src/portable/raspberrypi/rp2040/rp2040_usb.c", line 219, function: sync_ep_buffer
```

rppicomidi, 2022-11-22:
@DatanoiseTV do you see the same issue if you try the [midi2usbhub](https://github.com/rppicomidi/midi2usbhub) project? The same hardware should work, except now you have to use the CLI to route MIDI. It uses the latest MIDI host stack. I was able to
use an Arturia Keylab Essential 88 with it; it has 2 virtual cables.

DatanoiseTV, 2022-11-22:
@rppicomidi I didn't check that yet. What are the major differences between the USB MIDI Host implementation? I can give it a go tomorrow (and also send you some debug logs with increased verbosity, if that helps).

rppicomidi, 2022-12-12:
OK Thanks. I will close this issue. If you can describe what happens when replugging in a separate issue, I would appreciate it.

closed 2022-12-12 by rppicomidi