rppicomidi/midi2usbhost #8: Capturing incoming midi.

Terumi, 2023-01-16:
Hello, I'm sorry if this is not the right place to ask.
I get confused by your code.
How can I printf whatever comes from my midi device?
Does this project captures usb midi and fowards it to midi din or vise-versa?
Thank you!

rppicomidi, 2023-01-17:
This is not a known issue. Thank you for reporting the bug. Please tell me what is the git hash for midi2usbhost, the tinyusb library, and the pico-sdk. I have the Arturia Keylab Essential  88 keyboard so maybe I can reproduce it.

Terumi, 2023-01-17:
I found it:
I had a printf here:
if (status & UART_UARTMIS_TXMIS_BITS)
    {
        // ready to transmit
        uint8_t val;
        RING_BUFFER_SIZE_TYPE npopped = ring_buffer_pop_unsafe(&midi_uart->midi_uart_tx, &val, 1);
        if (npopped == 0)
        {
            // No data to send; disable the interrupt
            uart_set_irq_enables(midi_uart->midi_uart, true, false);
        }
        else
        {
            // send the byte
            **//printf(val + "\r\n");**
            uart_get_hw(midi_uart->midi_uart)->dr = val;
        }
    }

and that was ruining the program.
Sorry about that :)

closed 2023-01-17 by Terumi

rppicomidi, 2023-01-17:
@Terumi No problem. Please do report any other issues if you find them or request help here if you need it.