rppicomidi/midi2usbhost #10: Add CV and Gate

JustBKen, 2023-03-30:
A lot of Synths and all modular systems use a 1V/octave control voltage for pitch, and a gate signal (5V low/high = note on/off). It would be great to be able to use this project to control such synths or add CV/Gate outputs to MIDI keyboards. This can be done by PWM and filters or a DAC such as the MCP4725 (I2C) for a better signal. in general, a semitone is  83.33 milivolts (0.08333 V)

rppicomidi, 2023-03-31:
@JustBKen That is a good idea for a project, but it is out of scope for this one. The intent of this project is to do just 5-pin DIN MIDI to USB Host so it is easier to understand and to build. MIDI to CV conversion is on my list of projects I want to do. I will get to it one day. I hope this project is still useful to you. Many analog synthesizers also have a MIDI IN connector, and some will even convert that MIDI IN to a CV and Gate output.

JustBKen, 2023-03-31:
I guess mine is a nieche case. I am currently building a synth around a monotron duo inspired by the ms-20. Since the monotrons only have CV/Gate and no MIDI, i want to add a method to use my MIDI keyboard on it without using 2 microprocessors. Your project is the only one I found acting as a USB-MIDI host.  Do you have any tipps on how I could modify your code to achieve midi note to Pitch CV.

rppicomidi, 2023-04-03:
@JustBKen Sorry I didn't get back to you sooner. The application is cool. There a whole commercial projects that do what you are doing. I am glad you figured it out.

In my mind, it really shouldn't be a fork of this project called `midi2usbhost`. It is a different project altogether, perhaps `usb-midi-host2cv` or something like that. I might suggest that instead of just sharing your code with me, you should create a new repository and share it with everyone in your own github page. After all, you did the work. Post a link in this issue when it is ready. I will keep this issue open so I and others can find it.

JustBKen, 2023-04-05:
I shall do so, once I am finished and have fully tested it. Thank you for the permission and encouragement!

rppicomidi, 2023-04-10:
@JustBKen [This project](https://github.com/peterzimon/pico-midi2cv) looks cool and may be helpful to you.