rppicomidi/midi2usbhost #12: Multicable error

LenarOsx, 2023-09-17:
It works beautifully! but I'm getting an error when I connect a midi device that has two virtual cables, sometimes it detects the device after 10 secs then it unmount the device.

Is the a way to get it work with a multicable device? I have only tried with a keystation mk3 it has 2 virtual cables.
when I plug a simple 1 virtual cable it works great. 

Thanks

rppicomidi, 2023-09-18:
@LenarOsx I do not have a keystation mk3, but I do have an Arturia KeyLab Essential 88. It also has 2 virtual cables. I just re-tested it with the Arturia. It works as expected, so the issue is probably not the two virtual cables per se. Without more information, I cannot know what the issue is. Please attach to this issue any console output you see. If you have the software on your computer to dump the keystation mk3 USB descriptor, please attach a dump of the descriptor, too.

If it is disconnecting, and if it is not a software issue after all, it might also be a power issue or other electrical issue with your keyboard or the your hardware. If you can plug a powered hub between your Pico board USB port and your keyboard, that might help resolve the issue.

Finally, please be aware that, by design, `midi2usbhost` only routes MIDI data on USB MIDI cable 0 to and from the DIN MIDI ports. If the data you need is on another virtual cable, you may prefer to use [midi2usbhub](https://github.com/rppicomidi/midi2usbhub) with a multi-cable USB MIDI keyboard because you can use the CLI to direct MIDI data from each USB cable to the DIN MIDI port or not.

LenarOsx, 2023-09-19:
Thanks man, power issue like you said. Now it doesn't disconnect. But I can't get it to do what I want.

Basically I want to read data bytes from USB midi, and send it to another MCU esp32, to make an Bluetooth midi device. 

I've tried with the midi2usbhub, but it doesnt list the uart midi in and out. Do I have to connect an actual actual midi device to get it listed?

By the way, dip midi in and out is connected to pin 6 and 7(uart1) on the rp pi pico correct?

Thanks man, I'm pretty sure I'm gonna get the project done with your Code.

rppicomidi, 2023-09-20:
@LenarOsx You are correct that Pico board pins 6 (UART1 TX) and 7 (UART1 RX) are where the MIDI data come out. Please make sure you connect UART1 TX from the Pico board to the RX pin of your esp32 board, and vice versa. If your esp32 board has an open drain UART TX pin (as it should for MIDI), be sure to add a pull-up resistor to 3.3V on that pin. Because there is no way that I know of to make the RP2040 UART TX output open drain without resorting to implementing the UART transmitter using PIO (see [pio_midi_uart_lib](https://github.com/rppicomidi/pio_midi_uart_lib), for example), you won't need a pull-up resistor on the Pico board's UART1 TX pin.

The `list` command in the `midi2usbhub` project just showed the USB devices. It assumed you know the DIN MIDI port is there. That is a bug that I just fixed; it should show the UART MIDI port now. The `show` command shows a connection matrix of all the devices. You will see `MIDI-IN-A` and `MIDI-OUT-A` listed there.

All of this said, you can accomplish USB Host to Bluetooth MIDI with a single Pico W board if you use the [ble-midi2usbhost](https://github.com/rppicomidi/ble-midi2usbhost) project.

If you have enough information to solve your problem, please mark this issue closed.

LenarOsx, 2023-09-25:
It worked on my akai LPD8, thanks man. I could not make it work on my keystation 61. It listed all the ports, I tried with both and couldn't  get any output from Midi out tx. But it worked on my single cable midi device so that is amazing. I'm gonna  close the issue. Thanks!. Here is the descriptor dump if you wanna take a look. 
[DescriptorDump_Keystation_61_MK3.txt](https://github.com/rppicomidi/midi2usbhost/files/12718807/DescriptorDump_Keystation_61_MK3.txt)

closed 2023-09-25 by LenarOsx

rppicomidi, 2023-09-26:
@LenarOsx Thanks for the descriptor dump. It seems fine. I am baffled and sorry I couldn't help you to get it to work. If you want to dive deeper yourself, in the `midi2usbhost` project, take a look at the `CMakeLists.txt` file and uncomment the line
```
# set(LOG 2)
```
and change it to
```
# set(LOG 3)
```
and then make a new debug build.

The console output generated at line 189 of `usb_midi_host.c` might at least tell you if it is receiving MIDI data from your keyboard.