rppicomidi/midi2usbhost #14: Project based on RP2040, potential firmware target

DatanoiseTV, 2023-12-05:
Hey,

I am the guy behind the RP2040 based USB MIDI host and I am going to release it soon.
My plan is to open source the firmware, and I was thinking for compatibility reasons and community friendlyness,
it would be nice to base it on the midi2usbhost firmware.

[image]

@rppicomidi I am happy to send you a device if you would like one.
It is powered by USB-C, has TRS-A MIDI In and Out, a USB MUX, Power switch to external device via MOSFET and much more.

rppicomidi, 2023-12-08:
@DatanoiseTV Exciting news. I will leave this issue open for a while in case anyone will prefer your product to DIY.

rppicomidi, 2024-12-24:
@DatanoiseTV I hope you had a good year. I will close this issue today.

closed 2024-12-24 by rppicomidi