rppicomidi/midi2usbhost #15: Adafruit Feather with USB host

abarthch, 2024-04-11:
Considering that there is a Feather with integrated USB host, do you think it could be used for this project? This way there would be no need to have to use two Pico's... right?

https://thepihut.com/products/adafruit-feather-rp2040-with-usb-type-a-host

rppicomidi, 2024-04-17:
@abarthch There is currently no need for two Pico boards. The second Pico board in the photo is just my debug tool. In this project, because I am using a Picoprobe board for code download and serial port debug, I just hooked the VBus pin from my Picoprobe Pico board to the VBus pin of the board running this project.

Once you have programmed your Pico board, all you really need to do is attach an OTG adapter to the Pico USB port and provide 5V power to the VBus pin on the Pico. For example, what I do in the [midi2usbhub](https://github.com/rppicomidi/midi2usbhub) project is use a USB C breakout board to provide power from a phone charger.

If you want to use the Adafruit feather board, please look at the the examples for the [usb_midi_host](https://github.com/rppicomidi/usb_midi_host) driver code. They are not too different from this project.

Did this answer your question, or did I miss your point?

rppicomidi, 2024-06-13:
@abarthch My new project [midi2piousbhub](https://github.com/rppicomidi/midi2piousbhub) should directly support the [Adafruit Feather RP2040 with USB Type A Host](https://learn.adafruit.com/adafruit-feather-rp2040-with-usb-type-a-host) board. It also supports MIDI from a USB Host like a PC, Mac, or iPad. It is powered from the board's USB device port and the USB device port does double duty as a MIDI Device and a CDC Serial port (for the command line interface).

Can I close this issue?

abarthch, 2024-06-14:
Great! Looking forward to try this out once I get a board. Thank you for letting us know 💪

closed 2024-08-23 by rppicomidi

rppicomidi, 2024-08-23:
@aeonoea FWIW, I created in this project an Arduino sketch that uses the Adafruit Feather board to implement MIDI to USB Host function. Find the sketch in the Arduino directory under `midi2piousbhost`. I closed the issue. If that program or the `midi2piousbhub` project didn't fix this for you, please re-open it.

commit 45c59824 (2024-08-23): Fix #15: Create midi2piousbhost sketch