rppicomidi/midi2usbhost #16: Compilation error -- MIDI

copych, 2024-05-24:
@rppicomidi thanks for sharing your code!
I now have a problem compiling midi2usbhost. Ihave done the setup according to the README.md.
Arduino IDE 1.8.20
MIDI Library 5.0.2 (also tried 4.3.2)
Raspberry PI Pico/RP2040 package 3.8.1.
All other libraries -- latest versions available.
Generic RP2040 selected (tried Feather, Raspberry Pi Pico, but no luck)
I get the following error:
```
midi2usbhost:51:66: error: 'MidiHostSettings' was not declared in this scope
   51 | MidiInterface<SerialMIDI<HardwareSerial, DefaultSerialSettings>, MidiHostSettings> DINmidi((SerialMIDI<HardwareSerial, DefaultSerialSettings>&)serialDINmidi);
      |                                                                  ^~~~~~~~~~~~~~~~
```
Please, help with this issue.

copych, 2024-05-24:
I have found the reason. EZ_USB_MIDI_HOST has a lot of changes in 2.0.0. When I rolled back to 1.0.1, it compiled OK! Thanks.

rppicomidi, 2024-05-28:
@copych Sorry about that. I pushed changes. It should now work correctly with library EZ_USB_MIDI_HOST 2.0.0. If the code I pushed works for you, please close this bug. If it does not, please tell me what errors you are seeing.

copych, 2024-06-13:
Thank you, I've checked with the new 2.0.0 version of the library and it compiles and works as supposed!

closed 2024-06-13 by copych