rppicomidi/midi2usbhost #17: Request

Sugarbearscode, 2024-06-07:
Hi, 

building the executable t is proving a challenge as I have very little experience of C++ on the raspberry pico. 

can you upload the .UF2 file to GitHub as I am sure that I am not the only one that would just like to download it and try it without having to go through the build process. 

Great project btw.

rppicomidi, 2024-06-09:
@Sugarbearscode Thank you for your kind words. However, I do not post pre-built binaries.

You do not need C++ experience to build this code. You only need to install the pico-sdk according to the [getting started guide](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) and then follow the build instructions in this project's [README.md](https://github.com/rppicomidi/midi2usbhost/blob/main/C-Code/README.md) file.  If building any Raspberry Pi Pico C code is more than you want to learn how to do, perhaps you may find the [Arduino version of this project](https://github.com/rppicomidi/midi2usbhost/tree/main/arduino/midi2usbhost) easier to build. The Raspberry Pi support forums may also be helpful.

If you do decide to have a go at building the .UF2 file yourself and find it is not working, I am willing to help, but only if you describe what computer operating system you are using (Windows, Linux, MacOS), what steps you took to build the code, and attach output to show where things went wrong.

Sugarbearscode, 2024-06-11:
Ok. So I have got to the point of building the file but I get this error. 

I think the instructions are wrong for building on the pico (I dont have a linux machine but I get an error on "make .."when I use these instructions on the Raspberry PI 4

export PICO_SDK_PATH=$HOME/foo/pico-sdk/
cd $HOME/foo/midi2usbhost
mkdir build
cd build
cmake ..
make

should be (I think) this which is the standard (just wanted to make it easier. The midi2usbhost directory is the same level as the pico-examples directory.

export PICO_SDK_PATH=$HOME/foo/pico-sdk/
cd $HOME/**pico**/midi2usbhost/**C-Code/**
mkdir build
cd build
cmake ..
make

If I use this then I get as far as make, which works ok until it gets to Building C Object CMakeFiles/midi2usbhost.dir/midi2usbhost.c.obj

```
This is the output from make. 
home@raspberrypi4cm:~/pico/midi2usbhost/C-Code/build $ make
[  1%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  2%] Linking ASM executable bs2_default.elf
[  2%] Built target bs2_default
[  3%] Generating bs2_default.bin
[  4%] Generating bs2_default_padded_checksummed.S
[  4%] Built target bs2_default_padded_checksummed_asm
[  6%] Creating directories for 'ELF2UF2Build'
[  7%] No download step for 'ELF2UF2Build'
[  8%] No update step for 'ELF2UF2Build'
[  9%] No patch step for 'ELF2UF2Build'
[ 10%] Performing configure step for 'ELF2UF2Build'
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/home/pico/midi2usbhost/C-Code/build/elf2uf2
[ 12%] Performing build step for 'ELF2UF2Build'
[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.o
[100%] Linking CXX executable elf2uf2
[100%] Built target elf2uf2
[ 13%] No install step for 'ELF2UF2Build'
[ 14%] Completed 'ELF2UF2Build'
[ 14%] Built target ELF2UF2Build
[ 15%] Building C object CMakeFiles/midi2usbhost.dir/midi2usbhost.c.obj
/home/home/pico/midi2usbhost/C-Code/midi2usbhost.c:33:10: fatal error: bsp/board_api.h: No such file or directory
   33 | #include "bsp/board_api.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/midi2usbhost.dir/build.make:76: CMakeFiles/midi2usbhost.dir/midi2usbhost.c.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:1560: CMakeFiles/midi2usbhost.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
home@raspberrypi4cm:~/pico/midi2usbhost/C-Code/build $ if config

Can you suggest what is wrong?

Sugarbearscode, 2024-06-11:
requesting help

DatanoiseTV, 2024-06-11:
#if __has_include("bsp/board_api.h")
#include "bsp/board_api.h"
#else
#include "bsp/board.h"
#endif

rppicomidi, 2024-06-12:
@Sugarbearscode Thank you for posting this information. Clearly I need to update the instructions and I will do so shortly.

@DatanoiseTV I am not sure that the versions of TinyUSB that have the `bsp/board.h` file support Application USB Host drivers. I need to investigate.

Sugarbearscode, 2024-06-12:
@rppicomidi not sure what I did, but it seems to be fixed now. 

I went back over the steps of updating the TInysub library and it did have a reboot overnight, maybe a fresh pair of eyes have fixed it but it all compiled without any issues today. 

When I get time I will do everything from scratch again and see if I get the same issues. 

Thanks

rppicomidi, 2024-06-12:
@Sugarbearscode Wow. Thank you. Nevertheless, I am going to reopen this until I fix the documentation. @DatanoiseTV USB Host Application drivers were introduced August 15, 2023 and the file `bsp/board.h` was renamed to `bsp/board_api.h` August 3, 2023, so while I thank you for your suggestion, I am going to omit it.

commit 27e9d76a (2024-06-12): Fix #17: Update README.md
closed 2024-06-12 by rppicomidi