rppicomidi/midi2usbhost #19: How to run this project with Python (not C)

GalbertPi, 2024-11-01:
Hi there, first of all let me acknowledge the great work you did in this repo, it really looks you are an experienced programmer and importantly you are kind towards people looking for information and advice!
In my case, the point is I own a Raspberry Pi Zero 2 W and I need to send and receive MIDI messages by the means of Python scripts.
I already asked for support in the official [Adafruit forum](https://forums.adafruit.com/viewtopic.php?t=214469) but essentially I didn't receive resolutive help.
Could you please be so kind to help me understand how I could successfully connect my Raspberry to the Adafruit MIDI Featherwing, without any Feather board between, with the same UART connection you set up, without using the C language if possible?
I am a Delphi developer that learned object-oriented Python two years ago, and some Flutter for mobile apps, but I'm currently not able to reproduce your project as it is.
You may find all the relevant information about my situation and some sample scripts in the forum post I linked above, of course I am available to tell you more here as well.
Thank you very much in advance,
Fabio from Milan (IT)

rppicomidi, 2024-11-01:
@GalbertPi Fabio, I am afraid your question is out of scope for this project. I will try to point you in a direction to look. However, I do not have a solution to your problem.

Forgive me if you know all of the following. I don't know what kind of embedded system programming background you have. My understanding is that Delphi is a cross-platform application programming language based on Pascal. I used Turbo Pascal back in the day, but that was a long time ago.

The MIDI featherwing board has no intelligence. It has logic level to MIDI out current loop driver circuit and a MIDI IN opto-isolator current loop receiver to logic level circuit. To make the featherwing board work with any processor
- Make sure the UART I/O pins are wired to the featherwing board correctly
- Make sure the featherwing board's power and ground are correct.
- Make sure the CPU board UART is set to 31250 bits per second
- Make sure your software can read and write the UART correctly.

The Raspberry Pi Pico is based on a 133MHz dual core ARM Cortex M0 chip called the RP2040. The board costs $4 US. By reading and writing registers from software, you can program the correct serial port baud rate for MIDI (31250 bits per second) and directly write data to the UART and read data from the UART. Versions of Python that run on this board are able to run with no native operating system and often have special Python libraries for accessing the unique hardware.

The Raspberry Pi Zero 2 W is based on a 1GHz quad core Arm Cortex A53 Broadcom BCM2710A1 chip. It normally runs a version of the Linux OS and normally requires a device driver to access the UART. The board costs $15 US. I do not have experience programming the UART for this board. I do have a Pi Zero W board, but I have not done much with it. I have written a Python program or two, but I do not have any experience with Python on the Raspberry Pi. The [PySerial library](https://pythonhosted.org/pyserial/) looks like it might let you access the UART. However, you need to make sure it will let you set the baud rate to 31250 bits per second. If it wilI not do it, you will have to find a utility that will let you do it, or, if it is possible, write the driver yourself.

If you want your serial port MIDI to work with ALSA so the Linux `amidi` command works with it, that is more work to do if it has not already been done.

This looks like you will need to use your search engine a lot and maybe find help on the Raspberry Pi forums. I am afraid I cannot offer you help otherwise, especially as an issue for this project.

Good luck.

GalbertPi, 2024-11-02:
Hi there, thank you very much for providing some hints anyway!
I am uncertain about how to further troubleshoot my situation, maybe I shall somehow restart from zero, maybe I corrupted my current Raspberry image concerning serial communication, or try with a different board instead of the Adafruit model.
I'm just frustrated because the previous board from Domoshop, mentioned in the other forum post, run at the first attempt with the very same setup!
I may re-open the issue if I feel like communicating some updates, unless you have a website or other more appropriate ways to get in friendly touch.
Kind regards,
Fabio

rppicomidi, 2024-11-04:
@GalbertPi I happened to see this [video](https://www.youtube.com/watch?v=RbdNczYovHQ)
which directed me to this [GitHub repo](https://github.com/BlitzCityDIY/midiExperimentsWithRaspberryPi)
[This wiring diagram for MIDI IN](https://github.com/BlitzCityDIY/midiExperimentsWithRaspberryPi/blob/main/midiInCircuit_bb.png) is the same circuit that is on the MIDI Featherwing board. I think the example in the video uses the [py-midi](https://pypi.org/project/py-midi/) library

The info is for the Raspberry Pi 3 and 4, but perhaps also will work with Pi Zero W 2.

Please let me know if it works. If it does, please publish the example on GitHub and credit BlitzCityDIY for the help.

rppicomidi, 2024-11-04:
[This example](https://www.osaelectronics.com/learn/setting-up-raspberry-pi-for-midi/) uses the py-serial library and makes you code the MIDI bytes yourself.

GalbertPi, 2024-11-04:
Hi there, thank you very much for taking the time to make some research!
What you found may prove to be helpful in further investigation, and possibly solve my problem.
I will try something out when possible.
Kindly,
Fabio

rppicomidi, 2025-02-14:
@GalbertPi  I am closing this issue. I hope you find a project that will help you with Python.

closed 2025-02-14 by rppicomidi