rppicomidi/midi2usbhost #20: Attempting to import project into VSCode fails

doughadfield, 2025-06-28:
Hi, firstly thank you so much for putting this project on-line - it looks to be exactly what I'm looking for. However, when I attempt to import the project into my VSCode environment (on MacOs arm) I get the following failure:

...
[main] Configuring project: midi2usbhost 
[proc] Executing command: /opt/homebrew/bin/cmake -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/clang -DCMAKE_CXX_COMPILER:FILEPATH=/usr/bin/clang++ --no-warn-unused-cli -S/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/C-Code -B/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/build -G "Unix Makefiles" --debugger --debugger-pipe /tmp/cmake-debugger-pipe-5b5f5a88-c6a9-4b40-be85-8d206b6a71cb
[cmake] Not searching for unused variables given on the command line.
[cmake] Running with debugger on.
[cmake] Waiting for debugger client to connect...
[debugger] Connecting debugger on named pipe: "/tmp/cmake-debugger-pipe-5b5f5a88-c6a9-4b40-be85-8d206b6a71cb"
[cmake] Debugger client connected.
[cmake] PICO_SDK_PATH is /Users/doug.hadfield/.pico-sdk/sdk/2.1.1
[cmake] Target board (PICO_BOARD) is 'pico'.
[cmake] Using board configuration from /Users/doug.hadfield/.pico-sdk/sdk/2.1.1/src/boards/include/boards/pico.h
[cmake] Pico Platform (PICO_PLATFORM) is 'rp2040'.
[cmake] -- The C compiler identification is AppleClang 17.0.0.17000013
[cmake] -- The CXX compiler identification is AppleClang 17.0.0.17000013
[cmake] -- The ASM compiler identification is GNU
[cmake] -- Found assembler: /Users/doug.hadfield/.pico-sdk/toolchain/14_2_Rel1/bin/arm-none-eabi-gcc
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - failed
[cmake] -- Check for working C compiler: /usr/bin/clang
[cmake] -- Check for working C compiler: /usr/bin/clang - broken
[cmake] CMake Error at /opt/homebrew/share/cmake/Modules/CMakeTestCCompiler.cmake:67 (message):
[cmake]   The C compiler
[cmake] 
[cmake]     "/usr/bin/clang"
[cmake] 
[cmake]   is not able to compile a simple test program.
[cmake] 
[cmake]   It fails with the following output:
[cmake] 
[cmake]     Change Dir: '/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/build/CMakeFiles/CMakeScratch/TryCompile-KJEDSy'
[cmake]     
[cmake]     Run Build Command(s): /opt/homebrew/bin/cmake -E env VERBOSE=1 /opt/homebrew/bin/gmake -f Makefile cmTC_6f227/fast
[cmake]     /opt/homebrew/bin/gmake  -f CMakeFiles/cmTC_6f227.dir/build.make CMakeFiles/cmTC_6f227.dir/build
[cmake]     gmake[1]: Entering directory '/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/build/CMakeFiles/CMakeScratch/TryCompile-KJEDSy'
[cmake]     Building C object CMakeFiles/cmTC_6f227.dir/testCCompiler.c.o
[cmake]     /usr/bin/clang   -mcpu=cortex-m0plus -mthumb  -MD -MT CMakeFiles/cmTC_6f227.dir/testCCompiler.c.o -MF CMakeFiles/cmTC_6f227.dir/testCCompiler.c.o.d -o CMakeFiles/cmTC_6f227.dir/testCCompiler.c.o -c /Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/build/CMakeFiles/CMakeScratch/TryCompile-KJEDSy/testCCompiler.c
[cmake]     clang: error: unsupported argument 'cortex-m0plus' to option '-mcpu='
[cmake]     gmake[1]: *** [CMakeFiles/cmTC_6f227.dir/build.make:82: CMakeFiles/cmTC_6f227.dir/testCCompiler.c.o] Error 1
[cmake]     gmake[1]: Leaving directory '/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/build/CMakeFiles/CMakeScratch/TryCompile-KJEDSy'
[cmake]     gmake: *** [Makefile:134: cmTC_6f227/fast] Error 2
[cmake]     
[cmake]     
[cmake] 
[cmake] 
[cmake] 
[cmake]   CMake will not be able to correctly generate this project.
[cmake] Call Stack (most recent call first):
[cmake]   CMakeLists.txt:24 (project)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
[proc] The command: /opt/homebrew/bin/cmake -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_C_COMPILER:FILEPATH=/usr/bin/clang -DCMAKE_CXX_COMPILER:FILEPATH=/usr/bin/clang++ --no-warn-unused-cli -S/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/C-Code -B/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/build -G "Unix Makefiles" --debugger --debugger-pipe /tmp/cmake-debugger-pipe-5b5f5a88-c6a9-4b40-be85-8d206b6a71cb exited with code: 1
[proc] The command: pgrep -P 5811 exited with code: 1
[driver] Switching to kit: GCC 14.2.0 arm-none-eabi
[proc] Executing command: /opt/homebrew/bin/arm-none-eabi-gcc -v
[proc] The command: ninja --version failed with error: Error: spawn ninja ENOENT
[proc] The command: ninja-build --version failed with error: Error: spawn ninja-build ENOENT
[driver] Removing /Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/build/CMakeCache.txt
[driver] Removing /Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/build/CMakeFiles
[main] Configuring project: midi2usbhost 
[proc] Executing command: /opt/homebrew/bin/cmake -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_C_COMPILER:FILEPATH=/opt/homebrew/bin/arm-none-eabi-gcc -DCMAKE_CXX_COMPILER:FILEPATH=/opt/homebrew/bin/arm-none-eabi-g++ --no-warn-unused-cli -S/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/C-Code -B/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/build -G "Unix Makefiles"
[cmake] PICO_SDK_PATH is /Users/doug.hadfield/.pico-sdk/sdk/2.1.1
[cmake] Not searching for unused variables given on the command line.
[cmake] Target board (PICO_BOARD) is 'pico'.
[cmake] Using board configuration from /Users/doug.hadfield/.pico-sdk/sdk/2.1.1/src/boards/include/boards/pico.h
[cmake] Defaulting platform (PICO_PLATFORM) to 'rp2040' based on PICO_BOARD setting.
[cmake] Defaulting compiler (PICO_COMPILER) to 'pico_arm_cortex_m0plus_gcc' since not specified.
[cmake] Configuring toolchain based on PICO_COMPILER 'pico_arm_cortex_m0plus_gcc'
[cmake] Defaulting PICO_GCC_TRIPLE to 'arm-none-eabi'
[cmake] -- The C compiler identification is GNU 14.2.0
[cmake] -- The CXX compiler identification is GNU 14.2.0
[cmake] -- The ASM compiler identification is GNU
[cmake] -- Found assembler: /Users/doug.hadfield/.pico-sdk/toolchain/14_2_Rel1/bin/arm-none-eabi-gcc
[cmake] -- Detecting C compiler ABI info
[cmake] -- Detecting C compiler ABI info - done
[cmake] -- Check for working C compiler: /opt/homebrew/bin/arm-none-eabi-gcc - skipped
[cmake] -- Detecting C compile features
[cmake] -- Detecting C compile features - done
[cmake] -- Detecting CXX compiler ABI info
[cmake] -- Detecting CXX compiler ABI info - done
[cmake] -- Check for working CXX compiler: /opt/homebrew/bin/arm-none-eabi-g++ - skipped
[cmake] -- Detecting CXX compile features
[cmake] -- Detecting CXX compile features - done
[cmake] Build type is Debug
[cmake] Using regular optimized debug build (set PICO_DEOPTIMIZED_DEBUG=1 to de-optimize)
[cmake] -- Found Python3: /opt/homebrew/Frameworks/Python.framework/Versions/3.13/bin/python3.13 (found version "3.13.5") found components: Interpreter
[cmake] TinyUSB available at /Users/doug.hadfield/.pico-sdk/sdk/2.1.1/lib/tinyusb/src/portable/raspberrypi/rp2040; enabling build support for USB.
[cmake] Compiling TinyUSB with CFG_TUSB_DEBUG=1
[cmake] BTstack available at /Users/doug.hadfield/.pico-sdk/sdk/2.1.1/lib/btstack
[cmake] cyw43-driver available at /Users/doug.hadfield/.pico-sdk/sdk/2.1.1/lib/cyw43-driver
[cmake] lwIP available at /Users/doug.hadfield/.pico-sdk/sdk/2.1.1/lib/lwip
[cmake] mbedtls available at /Users/doug.hadfield/.pico-sdk/sdk/2.1.1/lib/mbedtls
[cmake] CMake Error at CMakeLists.txt:29 (add_subdirectory):
[cmake]   The source directory
[cmake] 
[cmake]     /Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/C-Code/lib/midi_uart_lib
[cmake] 
[cmake]   does not contain a CMakeLists.txt file.
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
[proc] The command: /opt/homebrew/bin/cmake -DCMAKE_BUILD_TYPE:STRING=Debug -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_C_COMPILER:FILEPATH=/opt/homebrew/bin/arm-none-eabi-gcc -DCMAKE_CXX_COMPILER:FILEPATH=/opt/homebrew/bin/arm-none-eabi-g++ --no-warn-unused-cli -S/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/C-Code -B/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/build -G "Unix Makefiles" exited with code: 1
...

I have pico-sdk 2.1.1 installed and have performed the "git pull" in your README (reported as already latest version). 

The above log shows two attempts: the first with the Clang toolchain and the second with the GCC toolchain.

What am I doing wrong?
cheers
Doug

rppicomidi, 2025-06-30:
@doughadfield  have no idea about using the CLANG toolchain for ARM on a Mac. I have never used it. A while back, I had an Intel Mac mini running HomeBrew and did command line builds. I used GCC.

From your gcc log, it looks like there are files missing. Please verify that `/Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/C-Code/lib/midi_uart_lib` contains the file `CMakeLists.txt`. If it does, please seek help from Mac VS Code users. I am afraid I cannot help you there.

If it does not, my guess is that either you did not use the `--recurse-submodules` option when you cloned the project, or it did not work on your Mac the same way it works under Linux. Since you have already cloned the project, try this:
```
cd /Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/C-Code/
git submodule update --recursive --init
```
Verify that /Users/Shared/MEGA/Guitar/MIDI/midi2usbhost/C-Code/lib/midi_uart_lib/CMakeLists.txt` exists after that command. If it does, retry your build. That should fix it.

doughadfield, 2025-06-30:
Thank you so much for your quick and full reply! - I'll work through your suggestions and see what I can do. I'm a C programmer of old and have a few pico projects under my belt (see github/doughadfield), but I confess I'm still fairly new to VisualStudioCode and the Mac toolchains (I'm more comfortable with Linux). I may also try VSC on Linux, as that may be a closer toolchain to the one you used. Thanks again - I'll let you know how I get on.

closed 2025-07-01 by doughadfield